adelolmo/hd-idle #121: spindown causing usb flash drive to fail

axet, 2024-11-10:
Hello!

```
hd-idle -i 1 -d
...
sda spindown
disk=sda command=scsi spunDown=true reads=1139234 writes=0 idleTime=1 idleDuration=60 spindown=2024-11-10T09:17:10 spinup=2024-11-10T09:16:10 lastIO=2024-11-10T09:16:10
sdb spindown
disk=sdb command=scsi spunDown=true reads=49992 writes=0 idleTime=1 idleDuration=60 spindown=2024-11-10T09:17:10 spinup=2024-11-10T09:16:10 lastIO=2024-11-10T09:16:10
sde spindown
cannot spindown scsi disk /dev/sde:
open /dev/sde: no medium found
...
```

```
[  861.883267] sdb: detected capacity change from 244328448 to 0
[  891.304235] sd 8:0:0:0: [sdb] tag#0 access beyond end of device
[  891.304239] I/O error, dev sdb, sector 1418752 op 0x0:(READ) flags 0x1000 phys_seg 3 prio class 0
[  891.304242] BTRFS error (device dm-2): bdev /dev/mapper/luks-2b9a7550-4a28-4a26-9bfd-5ce55a1589a0 errs: wr 0, rd 3, flush 0, corrupt 0, gen 0
[  891.304256] sd 8:0:0:0: [sdb] tag#0 access beyond end of device
[  891.304257] I/O error, dev sdb, sector 3515904 op 0x0:(READ) flags 0x1000 phys_seg 3 prio class 2
[  891.304258] BTRFS error (device dm-2): bdev /dev/mapper/luks-2b9a7550-4a28-4a26-9bfd-5ce55a1589a0 errs: wr 0, rd 4, flush 0, corrupt 0, gen 0
[  891.305453] sd 8:0:0:0: [sdb] tag#0 access beyond end of device
[  891.305457] I/O error, dev sdb, sector 280608 op 0x0:(READ) flags 0x1000 phys_seg 3 prio class 0
[  891.305462] BTRFS error (device dm-2): bdev /dev/mapper/luks-2b9a7550-4a28-4a26-9bfd-5ce55a1589a0 errs: wr 0, rd 5, flush 0, corrupt 0, gen 0
[  891.305483] sd 8:0:0:0: [sdb] tag#0 access beyond end of device
[  891.305484] I/O error, dev sdb, sector 2377760 op 0x0:(READ) flags 0x1000 phys_seg 3 prio class 2
[  891.305486] BTRFS error (device dm-2): bdev /dev/mapper/luks-2b9a7550-4a28-4a26-9bfd-5ce55a1589a0 errs: wr 0, rd 6, flush 0, corrupt 0, gen 0
~

```
lsusb
Bus 001 Device 006: ID aaaa:8816 MXT microSD CardReader
```

Can we have exclude list command line option?

adelolmo, 2024-11-10:
Hi @axet 
Please have a look to the [configuration section](https://github.com/adelolmo/hd-idle?tab=readme-ov-file#configuration), specially to the first example: `hd-idle -i 0 -a sda -i 300 -a sdb -i 1200`.

axet, 2024-11-10:
I did. I wasn't clear if using '-a' option limit drives to selected or only applied to following options '-i' etc...

`

axet, 2024-11-11:
hd-idle only need to spindown 'rotational' devices, which can be detected by the following property:

```
/sys/devices/pci0000:00/0000:00:02.4/0000:02:00.0/nvme/nvme0/nvme0n1/queue/rotational
/sys/devices/virtual/block/zram0/queue/rotational
```

adelolmo, 2024-11-11:
Hi @axet 
In your case you could just spin down the disks you want:
`hd-idle -i 0 -a sda -i 1 -a sdb -i 1`

closed 2024-12-17 by adelolmo