adelolmo/hd-idle #123: hd-idle only works once after a reboot

Conan179, 2024-11-24:
Hello. For me hd-idle only works once after a restart, then unfortunately no longer.
My /etc/default/hd-idle looks like this:
START_HD_IDLE=true
HD_IDLE_OPTS="-a ata-WDC_WD140EDGZ-11B1PA0_Y6GXK69C -i 600 -l /home/conan/hd-idle.log2"
My specified log file also remains empty.
root@debian_one:~# systemctl status hd-idle
● hd-idle.service - hd-idle - spin down idle hard disks
Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; preset: enabled)
Active: active (running) since Sun 2024-11-24 09:30:27 CET; 1h 18min ago
 Docs: man:hd-idle(1)
 Process: 551 ExecStart=/usr/sbin/hd-idle $HD_IDLE_OPTS (code=exited, status=0/SUCCESS)
 Main PID: 555 (hd-idle)
 Tasks: 1 (limit: 1779)
 CPU: 56ms
 CGroup: /system.slice/hd-idle.service
 └─555 /usr/sbin/hd-idle -a ata-WDC_WD140EDGZ-11B1PA0_Y6GXK69C -i 600 -l /home/toby/hd-idle.log2

Nov 24 09:30:26 debian_one systemd[1]: Starting hd-idle.service - hd-idle - spin down idle hard disks...
Nov 24 09:30:27 debian_one systemd[1]: Started hd-idle.service - hd-idle - spin down idle hard disks.

According to this the daemon seems to be running, am I doing something wrong?

adelolmo, 2024-12-17:
Hi @Conan179 

If yours is a scsi device, you should have a look to the parameter `-p power_condition`.
Otherwise you can use the parameter `-c command_type`.

Both parameters are explain in the [configuration section](https://github.com/adelolmo/hd-idle?tab=readme-ov-file#configuration).

Conan179, 2025-01-05:
No, SATA Hdds in a Qnap nas with Debian as os.

Conan179, 2025-01-05:
Ok, one of my problems is that the CPU of my NAS is a wreck. I can't install the deb package, the version of the package manager is "hd-idle_1.05+ds-2+b1_armel.deb" (doesn't recognize -c). When compiling, the error message appears:
Makefile:31: *** Invalid ARCH: .  Schluss.

adelolmo, 2025-01-05:
You could start by trying to install the [armhf](https://github.com/adelolmo/hd-idle/releases/download/v1.21/hd-idle_1.21_armhf.deb) version.

If that doesn't work, Debian `trixie` includes the packages for the latest version of hd-idle (note that I have no control of what is packaged for Debian).
It seems `trixie` offers a `armel` that might be useful in your case.
https://packages.debian.org/trixie/admin/hd-idle

Currently I don't have plans to offer packages on other platforms than the ones under releases: `amd64`, `arm64`, `armhf` and `i386`.

Conan179, 2025-01-05:
Unfortunately the installation of the armhf version does not work:
apt install "/home/con/hd-idle_1.21_armhf.deb"
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Hinweis: »hd-idle:armhf« wird an Stelle von »/home/con/hd-idle_1.21_armhf .deb« gewählt.
Die folgenden Pakete werden ENTFERNT:
  hd-idle
Die folgenden NEUEN Pakete werden installiert:
  hd-idle:armhf
0 aktualisiert, 1 neu installiert, 1 zu entfernen und 0 nicht aktualisiert.
Es müssen noch 0 B von 1.268 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 2.540 kB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] j
Holen:1 /home/con/hd-idle_1.21_armhf .deb hd-idle armhf 1.21 [1.268 kB]
dpkg: Fehler beim Bearbeiten des Archivs /home/con/hd-idle_1.21_armhf .deb (--unpack):
 Paket-Architektur (armhf) passt nicht zum System (armel)
Fehler traten auf beim Bearbeiten von:
 /home/con/hd-idle_1.21_armhf .deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

I'll try the Trixie version.

No thing.

Conan179, 2025-01-07:
I found out something. I worked with the -d option and I found out the following.
When the HDD is woken up by the Linux program hdsentinelarm (https://www.hdsentinel.com/how_to_monitor_network_attached_storage_nas_status.php) the debug log says that the HDD (sdc in my case) is not active. Does that help?
2025-01-07T21:27:14.456898+01:00 Qnap-NAS-eth0 hd-idle[507]: disk=sdc command=scsi spunDown=true reads=4176 writes=0 idleTime=600 idleDuration=3423 spindown=2025-01-07T20:40:11 spinup=2025-01-07T20:30:11 lastIO=2025-01-07T20:30:11