adelolmo/hd-idle #125: Is it possible to add a parameter to execute actions based on specific time periods?

bsdcpp, 2024-12-29:
I want certain drives that are accessed frequently during the day to skip sleep or have a longer idle interval, while allowing them to sleep normally at night.

Currently, the solution I have in mind is to schedule changes to the /etc/default/hd-idle configuration file and then restart the hd-idle service. However, this approach seems to result in the loss of some log records.

I'm not sure if this requirement is reasonable.

thank you for your effort.

adelolmo, 2025-01-05:
> However, this approach seems to result in the loss of some log records

Could you explain what logs are missing? Maybe we could solve this issue before jumping into changing code :)

bsdcpp, 2025-01-10:
> > However, this approach seems to result in the loss of some log records
> 
> Could you explain what logs are missing? Maybe we could solve this issue before jumping into changing code :)

I was mistaken earlier. I thought logs could only be viewed with `systemctl status hd-idle.service`, but actually, all logs can be viewed using `journalctl -u hd-idle.service`.

To achieve the goal mentioned in the title, there are currently two methods:  
1. Regularly modify the configuration and restart the service.  
2. For hard drives that do not need to hibernate, perform periodic access/read-write operations.

adelolmo, 2025-01-25:
> 2. For hard drives that do not need to hibernate, perform periodic access/read-write operations.

Or just simply exclude them from your `/etc/default/hd-idle` [configuration](https://github.com/adelolmo/hd-idle?tab=readme-ov-file#configuration).

closed 2025-01-25 by adelolmo