adelolmo/hd-idle #126: Extended offline self-test (smartctl -t long) aborts due to hd-idle

presto8, 2025-02-24:
If hd-idle is running, extended offline self-tests will, after some time, get cancelled with the reason "Aborted by host." If I disable hd-idle, the extended offline test completes successfully. 

Is there any way to keep hd-idle from idling a drive if a self-test is running?

For a long time, I used "while sleep 60s; do dd if=/dev/sda iflag=direct count=1 of=/dev/null; done" to keep the drive awake but doesn't work any more.

As a note: On larger drives, like 14T or greater, the extended offline test can take over 24 hours to complete.

Thank you!

adelolmo, 2025-03-07:
Hi @presto8 

You could configure hd-idle not to spin down disk by default (`-i 0`) and remove or add disks (option `-a`) depending on if the self-test is running or not.

presto8, 2025-03-09:
@adelolmo, thank you! I will give it a try!

What I also tried was the following which seems to work. Since I manage hd-idle with systemctl:

```
trap "systemctl start hd-idle" EXIT
systemctl stop hd-idle
while sleep 5m; do
    if ! smartctl -a /dev/sda | grep 'of test remaining'; then
        break
    fi
done
```