adelolmo/hd-idle #129: check wether disk is spin up or down?

DocMAX, 2025-08-13:
is there a way to get the status of a drive where it is idle or not?
the problem is i have a usb hd dock with a jmicron chipset. hd-idle -t seems to work.
but i can't find a way to get the status. smartctl always says active even if idle.
and ata command do spinup the disk for query.

DocMAX, 2025-08-13:
somehow hd-idle "detects" that disk is spinup. can you create some file with the current status somewhere? like /sys/block/sda/power/status (status is "up" or "down" for example). this way i can poll for other scripts to stop if disk is down. or any better ideas how to do it?

dallmair, 2025-08-16:
You can use [hdparm](https://wiki.archlinux.org/title/Hdparm) for that, e.g. `hdparm -C /dev/sda`

```

/dev/sda:
 drive state is:  standby
```

An alternative is [smartctl](https://wiki.archlinux.org/title/Hdparm#Querying_the_status_of_the_disk_without_waking_it_up), e.g. `smartctl -i -n standby /dev/sda`:

```
smartctl 7.4 2023-08-01 r5530 [x86_64-linux-6.12.41+deb13-amd64] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in STANDBY mode, exit(2)
```

DocMAX, 2025-08-16:
smartctl does that too, as it's using SAT commands. (i am using a USB enclosure as i mentioned)

xorrvin, 2026-01-17:
here's what works for me in a similar situation (spinned down SAS drive, hdparm or smartctl wakes it up):
```
# sdparm -v --command=sense /dev/sdg
    /dev/sdg: SEAGATE   ST2000NX0273      E005
    request sense cdb: [03 00 00 00 40 00]
Decode response as sense data:
Fixed format, current; Sense key: No Sense
Additional sense: Standby condition activated by command
Additional sense: Standby condition activated by command
```

one can grep for "additional sense" or "standby"; if drive is active there are no such lines in the output.