adelolmo/hd-idle #131: Regression in v1.21: Drives fail to spin down after the first wake-up cycle

nicoh88, 2025-08-29:
### Environment
**hd-idle (working):** 1.05
**hd-idle (faulty):** 1.21+ds-1+b8 (from the Debian 13 "Trixie" repository)
**Host OS:** Proxmox VE 9 (based on Debian 13)
**Setup Specifics:** Three SATA HDDs are passed through to a Debian VM using "Raw Disk Passthrough". hd-idle is running on the Proxmox host, as the guest VM cannot control the physical power state of the drives.    

### Bug Description
A specific regression is present in `hd-idle` version 1.21. After restarting the `hd-idle` service, the drives are correctly put into standby after the specified timeout. However, once the drives are woken up by any access, `hd-idle` fails to put them back into standby after they become idle again.

The core issue appears to be that the daemon does not correctly recognize or process the new "active" state of the disk after the first wake-up event. Consequently, the idle timer is not reset, and no further spindown commands are sent.

This behavior does not occur with the older version 1.05, which handles repeated spin-down cycles reliably under the exact same configuration.

### Expected Behavior
After a disk wakes up, `hd-idle` should detect the activity, reset its idle timer, and successfully spin the disk down again after the timeout period expires.

**Actual Behavior**
`hd-idle` only spins the disks down once after the service starts. After the first wake-up, the disks remain permanently in the `active/idle` state, as confirmed by `hdparm -C`. A `service hd-idle restart` is required to trigger the spin-down cycle one more time.

### Additional Context and Troubleshooting
**Workaround:** Downgrading to `hd-idle` version 1.05 completely resolves the issue.
**Troubleshooting Attempt:** Adding the `-c ata` parameter to the `HD_IDLE_OPTS `in version 1.21 was attempted but had no effect. The behavior remained unchanged.
**Related Issue:** This problem appears to be directly related to Issue #123 , which describes a similar state-tracking problem.

### Configuration and Logs
**/etc/default/hd-idle Configuration:**
```
START_HD_IDLE=true
# 15 Minuten
# HD_IDLE_OPTS="-l /var/log/hd-idle.log -i 0 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAKKATYL -i 900 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAG81VEL -i 900 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VDHLMYRD -i 900"
# 1 Minute
#HD_IDLE_OPTS="-l /var/log/hd-idle.log -i 0 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAKKATYL -i 60 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAG81VEL -i 60 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VDHLMYRD -i 60"
# 1 Minuten inkl. "ata"
#HD_IDLE_OPTS="-c ata -l /var/log/hd-idle.log -i 0 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAKKATYL -i 60 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAG81VEL -i 60 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VDHLMYRD -i 60"
```

**`journalctl -u hd-idle` (shows the first, successful cycle):**

```
Aug 29 08:36:29 proxmox hd-idle[562610]: /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAG81VEL spindown
Aug 29 08:36:31 proxmox hd-idle[562610]: /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAKKATYL spindown
Aug 29 08:36:32 proxmox hd-idle[562610]: /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VDHLMYRD spindown
```

*(Note: After this initial cycle, no further `spindown` or `spinuo` events are logged, even though the disks remain active).*

**`/etc/lvm/lvm.conf` (proxmox additional):**
```
devices {
     # added by pve-manager to avoid scanning ZFS zvols and Ceph rbds
     #global_filter=["r|/dev/zd.*|","r|/dev/rbd.*|"]
     global_filter=["r|/dev/zd.*|","r|/dev/rbd.*|", "r|/dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAKKATYL.*|", "r|/dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAG81VEL.*|", "r|/dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VDHLMYRD.*|"]
}
```

adelolmo, 2025-08-30:
Hi @nicoh88 
First of all, thank you for your detail report.
Would you mind starting the service in debug mode? Just add the flag `-d` to the `HD_IDLE_OPTS`.
This will generate a lot of entries, so I don't recommend it leaving it on for a long time.
Thank you

nicoh88, 2025-09-01:
> Hi [@nicoh88](https://github.com/nicoh88) First of all, thank you for your detailed report. 

👍 

> Would you mind starting the service in debug mode? Just add the flag `-d` to the `HD_IDLE_OPTS`. This will generate a lot of entries, so I don't recommend it leaving it on for a long time. 

Sure. 💯 

### 1.05 (bookworm repo)

```
hd-idle -d -i 0 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAKKATYL -i 60 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAG81VEL -i 60 -a /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VDHLMYRD -i 60
using sdc for /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAKKATYL
using sdb for /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VAG81VEL
using sda for /dev/disk/by-id/ata-WDC_WD80EFAX-68KNBN0_VDHLMYRD
probing sda: reads: 57796796, writes: 68574683
probing sdb: reads: 58230355, writes: 68509523
probing sdc: reads: 57826901, writes: 68432363
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57806852, writes: 68581070
probing sdb: reads: 58239563, writes: 68516110
probing sdc: reads: 57836541, writes: 68438934
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57809684, writes: 68582040
probing sdb: reads: 58242219, writes: 68517712
probing sdc: reads: 57839677, writes: 68440536
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57809684, writes: 68582048
probing sdb: reads: 58242219, writes: 68517720
probing sdc: reads: 57839677, writes: 68440544
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586298
probing sdb: reads: 58244283, writes: 68521986
probing sdc: reads: 57841701, writes: 68444850
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586306
probing sdb: reads: 58244283, writes: 68521994
probing sdc: reads: 57841701, writes: 68444858
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586306
probing sdb: reads: 58244283, writes: 68521994
probing sdc: reads: 57841701, writes: 68444858
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586314
probing sdb: reads: 58244283, writes: 68522002
probing sdc: reads: 57841701, writes: 68444866
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586314
probing sdb: reads: 58244283, writes: 68522002
probing sdc: reads: 57841701, writes: 68444866
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586314
probing sdb: reads: 58244283, writes: 68522002
probing sdc: reads: 57841701, writes: 68444866
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586314
probing sdb: reads: 58244283, writes: 68522002
probing sdc: reads: 57841701, writes: 68444866
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586314
probing sdb: reads: 58244283, writes: 68522002
probing sdc: reads: 57841701, writes: 68444866
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586314
probing sdb: reads: 58244283, writes: 68522002
probing sdc: reads: 57841701, writes: 68444866
probing sdd: reads: 49901, writes: 393831033
probing sda: reads: 57811764, writes: 68586314
probing sdb: reads: 58244283, writes: 68522002
probing sdc: reads: 57841701, writes: 68444866
[... 318 lines cut ...]
disk=sda command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=260 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdb command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=260 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdc command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=260 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdd command=scsi spunDown=false reads=40930 writes=393831033 idleTime=0 idleDuration=260 spindown=0001-01-01T00:00:00 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sda command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=266 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdb command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=266 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdc command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=266 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdd command=scsi spunDown=false reads=40930 writes=393831033 idleTime=0 idleDuration=266 spindown=0001-01-01T00:00:00 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdd command=scsi spunDown=false reads=40930 writes=393831033 idleTime=0 idleDuration=272 spindown=0001-01-01T00:00:00 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sda command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=272 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdb command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=272 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdc command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=272 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sda command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=278 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
[... cut ...]
disk=sdc command=scsi spunDown=true reads=32 writes=0 idleTime=60 idleDuration=284 spindown=2025-09-01T09:55:46 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
disk=sdd command=scsi spunDown=false reads=40930 writes=393831033 idleTime=0 idleDuration=284 spindown=0001-01-01T00:00:00 spinup=2025-09-01T09:54:46 lastIO=2025-09-01T09:54:46
```

It's not detecting any `reads `or `writes`, which means `lastIO` isn't working either. That seems to be the issue, but why? What changed in version 1.05 and, more importantly, why isn't it affecting everyone?

Thanks for fixing @adelolmo. 🥇

wumpus71, 2025-11-03:
Hi adelolmo,

> I have some ideas, but I'd like to hear your opinions first of all.

For me, it is a don't care. It seems that there are really not many complains about tracking partition stats, so I seems that the majority of the uses go well with the partition variant, so I would leave this as default.
Just, don't make it too complicated e.g. by guessing what variant could be correct. Then, _something_ changes, causing the algorithm to fail w/o apparent cause, and leaving the puzzled user alone in heavy rain ... ;-).

ambiman, 2025-11-13:
I'm also facing the same issues with PVE on Debian 13 and would appreciate an option to make it configurable 👍

adelolmo, 2026-01-09:
Hi @wumpus71 

This [pre-release](https://github.com/adelolmo/hd-idle/releases/tag/v1.22) removes partitions from the disk activity calculation. Could you please give it a try and share how it went?
It would be amazing if you could also record the activity with https://github.com/adelolmo/hd-idle-test :)

wumpus71, 2026-01-11:
Hi again,

I got the hd-ide-test compiled & installed sucessfully, however hdt did not work as expected. I started hd-idle as specified in the instructions. with the same file names provided, and I was able to start the recording with ctrl-r. However, when I wanted to load a session, it either crashed (see below), or it just stated "failed to load session".
I however realized that I used go 1.24.4, the newest is 1.25.5, I will give that a try in the next days.

hdt crash log:
```
root@asgard:~# hdt
panic: runtime error: index out of range [-1] [recovered]
        panic: runtime error: index out of range [-1]

goroutine 1 [running]:
github.com/rivo/tview.(*Application).Run.func1()
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/application.go:319 +0x45
panic({0x77ce00?, 0xc0000d4018?})
        /usr/lib/go-1.24/src/runtime/panic.go:792 +0x132
main.main.func3(0xc000097700)
        /root/hd-idle/hd-idle-test/tui/main.go:194 +0x50a
github.com/rivo/tview.(*Flex).InputHandler.(*Box).WrapInputHandler.func2(0xc00010bb00?, 0xc0003a47d0)
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/box.go:164 +0x37
github.com/rivo/tview.(*Flex).InputHandler.func1(0xc000097700, 0xc0003a47d0)
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/flex.go:255 +0xd7
github.com/rivo/tview.(*Flex).InputHandler.(*Box).WrapInputHandler.func2(0xc00010bbf0?, 0x417185?)
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/box.go:167 +0x50
github.com/rivo/tview.(*Flex).InputHandler.func1(0xc000097700, 0xc0003a47d0)
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/flex.go:255 +0xd7
github.com/rivo/tview.(*Flex).InputHandler.(*Box).WrapInputHandler.func2(0xc00010bbc0?, 0x46eed9?)
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/box.go:167 +0x50
github.com/rivo/tview.(*Flex).InputHandler.func1(0xc000097700, 0xc0003a47d0)
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/flex.go:255 +0xd7
github.com/rivo/tview.(*Flex).InputHandler.(*Box).WrapInputHandler.func2(0xc00010bd40?, 0xc000027bc0?)
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/box.go:167 +0x50
github.com/rivo/tview.(*Application).Run(0xc00017a000)
        /root/hd-idle/hd-idle-test/tui/vendor/github.com/rivo/tview/application.go:441 +0x962
main.main()
        /root/hd-idle/hd-idle-test/tui/main.go:267 +0x38ed

```

Edit: just noticed that I got some messages from htdt in the system journal, which dont't seem to be too good:
```
Jan 12 00:18:19 asgard hdtd[3017939]: [GIN-debug] POST   /record                   --> main.main.func4 (3 handlers)
Jan 12 00:18:19 asgard hdtd[3017939]: [GIN-debug] GET    /status                   --> main.main.func3 (3 handlers)
Jan 12 00:18:19 asgard hdtd[3017939]: [GIN-debug] GET    /sessions/:id             --> main.main.func2 (3 handlers)
Jan 12 00:18:19 asgard hdtd[3017939]: [GIN-debug] GET    /sessions                 --> main.main.func1 (3 handlers)
Jan 12 00:18:19 asgard hdtd[3017939]:  - using code:        gin.SetMode(gin.ReleaseMode)
Jan 12 00:18:19 asgard hdtd[3017939]:  - using env:        export GIN_MODE=release
Jan 12 00:18:19 asgard hdtd[3017939]: [GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
Jan 12 00:18:19 asgard hdtd[3017939]: [GIN-debug] [WARNING] Creating an Engine instance with the Logger and Recovery middleware already attached.
Jan 12 00:18:19 asgard systemd[1]: Started hdtd.service - Help testing hd-idle for real usecase scenarios.
Jan 12 00:18:19 asgard systemd[1]: hdtd.service: Scheduled restart job, restart counter is at 3.
Jan 12 00:18:19 asgard systemd[1]: hdtd.service: Failed with result 'exit-code'.
Jan 12 00:18:19 asgard systemd[1]: hdtd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jan 12 00:18:19 asgard hdtd[3017138]:         /root/hd-idle/hd-idle-test/daemon/main.go:115 +0x587
Jan 12 00:18:19 asgard hdtd[3017138]: created by main.main.func2 in goroutine 50
Jan 12 00:18:19 asgard hdtd[3017138]:         /root/hd-idle/hd-idle-test/daemon/main.go:120 +0x1ec
Jan 12 00:18:19 asgard hdtd[3017138]: main.main.func2.2()
Jan 12 00:18:19 asgard hdtd[3017138]: goroutine 53 [running]:
Jan 12 00:18:19 asgard hdtd[3017138]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0xd1c80c]
Jan 12 00:18:19 asgard hdtd[3017138]: panic: runtime error: invalid memory address or nil pointer dereference
Jan 12 00:18:19 asgard hdtd[3017138]:         /root/hd-idle/hd-idle-test/daemon/main.go:124 +0x653
Jan 12 00:18:19 asgard hdtd[3017138]: created by main.main.func2 in goroutine 50
Jan 12 00:18:19 asgard hdtd[3017138]:         /root/hd-idle/hd-idle-test/daemon/main.go:129 +0x1ec
Jan 12 00:18:19 asgard hdtd[3017138]: main.main.func2.3()
Jan 12 00:18:19 asgard hdtd[3017138]: goroutine 54 [running]:
Jan 12 00:18:19 asgard hdtd[3017138]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0xd1c44c]
Jan 12 00:18:19 asgard hdtd[3017138]: panic: runtime error: invalid memory address or nil pointer dereference
Jan 12 00:18:19 asgard hdtd[3017138]: 2026/01/12 00:18:19 open /root/.config/hdtd/1768173186/1768173191/log: no such file or directory
Jan 12 00:18:19 asgard hdtd[3017138]: 2026/01/12 00:18:19 open /root/.config/hdtd/1768173186/1768173191/stdout: no such file or directory
Jan 12 00:18:14 asgard hdtd[3017138]: [GIN] 2026/01/12 - 00:18:14 | 200 |     116.282µs |                 | GET      "/status"
Jan 12 00:18:14 asgard hdtd[3017138]: [GIN] 2026/01/12 - 00:18:14 | 200 |      91.899µs |                 | GET      "/sessions"
Jan 12 00:18:07 asgard hdtd[3017138]: [GIN] 2026/01/12 - 00:18:07 | 200 |     133.277µs |                 | GET      "/status"
Jan 12 00:18:07 asgard hdtd[3017138]: [GIN] 2026/01/12 - 00:18:07 | 200 |     142.415µs |                 | GET      "/sessions"
```
Jan
Regards, Wumpus71

adelolmo, 2026-01-18:
Hi @wumpus71 

First of all, I have to thank you for your perseverance! I know it can get frustrating sometimes, and I'd like to apologize for the trouble.

I have made some changes to the tooling, so please go ahead, and pull the latest code from https://github.com/adelolmo/hd-idle-test and restart the daemon.
There are some stability improvements that will help us debugging your issue (and reduce frustration).

For instance, I killed the hd-idle process, removed the files `/tmp/hd-idle.out` and `/var/log/hd-idle.log` and create a new recoding. When I select the session, I get the following error message in the log bar:
`Error loading session. server error: open /root/.config/hdtd/1768726940/1768726945/log: no such file or directory`

If you get the same error, then you know that the files `/tmp/hd-idle.out` and `/var/log/hd-idle.log` are not created, because on session recording those files are read by the daemon to create the files `stdout` and `log` for each session snapshot (aka frame).

I hope the changes help us to focus on the actual problem :)

wumpus71, 2026-01-19:
Hi adelolmo,

no clue what happens, but `hd-idle` fails to generate the logfile, independent of the filename I provide with the `-l` option. The file just does not exist. If I do not provide a filename, `hd-idle` quits with an error message as it should.
I also downgraded to 1.21, with no success.

Regards, Wumpus

EduardoInTech, 2026-10-19:
Same symptom here on v1.21 (Debian 13), and I think I found the cause: the device-mapper holder substitution in `diskstats/snapshot.go`.

When a disk has holders, `readSnapshot()` **replaces** the disk's counters with those of the holder returned by `getDiskHolder()`, which only returns the **first** entry of `/sys/class/block/<disk>/holders/`. With LVM thin provisioning the disk has two holders, `_tmeta` and `_tdata`, and the first one (alphabetically) is `_tmeta`:

```
/sys/class/block/sdb/holders/ -> dm-4 (vg-tpool_tmeta), dm-5 (vg-tpool_tdata)
```

So hd-idle only sees the thin pool **metadata** I/O. Data reads go to `dm-5` and are invisible. Verified with `-d`:

```
disk=sdb ... reads=2944984 writes=388296      # what hd-idle sees
dm-4 sectRead=2944984 sectWritten=388296      # identical to dm-4
sdb  sectRead=492982200 sectWritten=771965032 # the real disk
```

Resulting behaviour, which matches this issue:
1. hd-idle spins the disk down and sets `SpunDown=true`.
2. Something reads data (a media server library scan in my case). The disk spins up, but `dm-4` doesn't change if the pool metadata is still cached.
3. Counters look unchanged, so hd-idle keeps believing the disk is asleep and never sends standby again. My disk stayed spinning for 28 h, until the weekly logrotate restart.

It only "works" when the wake-up also reads pool metadata, which is why it looks intermittent. The same applies to any disk whose first holder isn't the one carrying the I/O. Possible fixes: keep the raw disk counters (they already include all holder I/O), or sum all holders instead of taking the first one.

d4k0, 2026-10-19:
I've had the same problem ever since I recently upgraded from Ubuntu Server 24.04 LTS to 26.04 LTS. `hd-idle` managed to put the hard drive (a WD Elements 20 TB) into sleep mode when it first started, but then something kept waking it up, and it stayed on all the time. I also tried the 1.22 beta version, but it only worked with the new `-I` parameter - which, however, is a brute-force method.

`fatrace` and `blktrace` didn’t provide any clear results either. With `blktrace`, though, I noticed the following entries that appeared regularly (about every 20 minutes):

```
8,16  12        1     0.000000000 1466903  D   N 0 [pool-547]
8,16  17        1     7.144679507 1453098  C   N [65531]
```

I then used ChatGPT to find out that `udisks2` was the cause. It runs a housekeeping job every 10 minutes (apparently to query S.M.A.R.T. values); this interval is apparently hard-coded in the code, and the issue has been known for years. The `udisks2` log also regularly showed that the query for S.M.A.R.T. values had failed:

```
Okt 17 23:34:12 ubuntu-server udisksd[1591896]: Error performing housekeeping for drive /org/freedesktop/UDisks2/drives/WDC_WD200EDGZ_XXXXXXX_XXXXXXXX: Error updating SMART data: Error sending ATA command CHECK POWER MODE: Unexpected sense data returned:
                                                0000: 70 00 02 00  00 00 00 0a  00 00 00 00  04 01 00 00    p...............
                                                0010: 00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00    ................
                                                 (g-io-error-quark, 0)
```

**These S.M.A.R.T. queries also do not appear in `/proc/diskstats`, which is probably why `hd-idle` does not detect that the hard drive was started by the query.**

I then used a udev rule to exclude the hard drive from all checks; someone posted the rule here:

https://github.com/storaged-project/udisks/issues/892#issuecomment-4722309520

With this setting, the hard drive no longer wakes up constantly.

You can check whether `udisks2` is the problem by stopping it temporarily - for example, for an hour - to see what happens:

`sudo systemctl stop udisks2`

This appears to be a bug in `udisks2`, which seems to have been [fixed in version 2.11.1](https://github.com/storaged-project/udisks/issues/1443#issuecomment-4021518356). Unfortunately, Ubuntu 26.04 (and likely Debian 13 as well) is still using an older version. I’m not entirely sure why I didn’t encounter this problem on Ubuntu 24.04, but other users have also reported that this issue didn’t occur with older versions of `udisks2`.

Here are a few more links to issues on this topic:

[10 Minute housekeeping_timeout keeps Western Digital Drives awake. #892](https://github.com/storaged-project/udisks/issues/892)

[Preventing udisks2 from waking up SATA hard disks in an USB enclosure #1443](https://github.com/storaged-project/udisks/issues/1443)

[Error sending ATA command CHECK POWER MODE #965](https://github.com/storaged-project/udisks/issues/965)

[Udisks2 prevents HDDs from going to standby because of polling SMART information](https://bugs.launchpad.net/ubuntu/+source/udisks2/+bug/1373318)